njfaria/dalylog #1: Post Code

dfense, 2020-10-23:
I really like the idea of pushing the daly BMS data over MQTT topic. 
Are you going to post code for this ? 
Are the screenshots of existing software posted somewhere else ? 

Hopeful the community can get access to this. Really gives a lot of 'connected' options for the Daly BMS product(s)

njfaria, 2020-10-23:
Software for raspberry is online under releases.
https://github.com/njfaria/dalylog/releases/tag/0.9.0
Just use ypur raspberry desktop..

dfense, 2020-10-25:
A bit confused, as all I see checked into the 0.9.0 tag branch (or any) are

- 2 PNG files
- 1 README.md
- 1 dummy.txt 
<no program or scripting code anywhere>

Am i missing something?
I downloaded the *.tar.gz and the *.zip plus browse the git online.

njfaria, 2020-10-26:
check the link above.. its a release, there is no source code.
its a zip file, which you unzip on your raspberry.

closed 2020-11-27 by njfaria