njfaria/dalylog #2: Os to run DalyLog

danielforster, 2021-02-06:
Hi,

What OS will DalyLog run on? I see that several people have it running on an RPi4, would that mean it runs on Raspbian Buster with Desktop?

Many thanks

Dan

njfaria, 2021-02-06:
They call the Operating system now Raspberry PI OS.
Dalylog runs on Raspberry 2,3,4 and should run on the zero too.

danielforster, 2021-02-07:
Wonderful, I shall try today on a Raspberry Pi Zero and come back with the outcome. Many thnaks for coming back so fast - Take care - Daniel

danielforster, 2021-02-08:
Nuno,

Can you give me some install details for installing DalyLog?

I have a Raspbery Pi 4 Model B (4GB) and also to try, a Raspberry Pi Zero. I have downloaded the latest Raspian image and burnt this to two speperate micro sd cards (one for the RPi4 and one for the RPI Zero). I have these booting all fine. I have also downloaded the DalyBmsMonitorV1.1.5 zip. Do I need to unzip this before installing on the Pi OS?

Many thanks in advance.

Daniel

njfaria, 2021-02-08:
Unzip it to the desktop ( for testing or prefarably create a folder named
dalylog and unzip it there) and double click on it... just like windows.
If you use the folder there will be also a setup.sh which you can double
click and execute..it should make the needed copy so that dalylog will be
autostart on reboot.
I hope my written english is understandable.

danielforster, 2021-02-09:
uno,

It all works fine :-). Very happy. I have gone with the RPi Zero which is very small and connects perfectly. My next  ask is can you give some detailed instructions on how to run on reboot, so it always run, as I am running with no monitor. As always, your expert help is much appreciated (and your English is just perfect by the way).

Take care,

Daniel

njfaria, 2021-02-09:
If you created the dalylog folder just execute the setup.sh
It will ask if you want to execute it in terminal. Just confirm.
After that it should autostart.
Just dont forget to tick autoload config and start on dalylog software and
click on save.

njfaria, 2021-02-10:
I just realized i didnt copy the setup.sh into the zip file..
Stay tuned for update

danielforster, 2021-02-10:
Nuno,

I did see that and was going to say something, but glad you have come back.

It all works really well on the Pi Zero. Interestingly I do note that only one of the mini USB sockets (with suitable adapter) works with the Daly USB-UART lead, the one more central to the Pi Zero, both work for power though, this stumped me for a while. I have made 4 Pi Zeros with the software you have coded, one for each of my 16 cell, 105 A 48V battery packs, and these are now reporting to MQTT. Today I will make up a nice dashboard for them all.

Again, a big thankyou for such excellent work, you are a clever person indeed.

Take care,

Dan

spilegi, 2021-02-11:
Is there any source code on this?  I'd love to see how you communicate with the BMS.

njfaria, 2021-02-11:
@danielforster 
i updated the release.. setup.sh is included
just unzip to dalylog folder
@spilegi 
no source for this, but if you are interested in the protocoll i will search for the link.

spilegi, 2021-02-11:
@njfaria Ok that would be cool  I'm having trouble interacting with the can bus, and was curious on how you did that.  I'm just looking to dump the cell values to a DB or even a text file.

spilegi, 2021-02-11:
I use a program called mbpoll to interact with modbus.  I was hoping there would be something similar for can, but your project seems to be the closest thing i see.  Sorry for the double post.

danielforster, 2021-02-11:
Cheers Nuno. I will sort in the morning and come back to confirm all is good.

Take care,

Dan

danielforster, 2021-02-12:
Nuno,

Just added the .sh file, I do note that it is zipped with NO EXECUTE command flag, so I had to right click and change to EXECUTE for it to then make the required changes. 

Have a lovely weekend.

Daniel

danielforster, 2021-02-16:
Nuno,

All going well. One question. My Dalylog RPi's report to my MQTT Server every 5 or 6 seconds (or so). Is it possible to reduce this rate to, say, every minute or couple of minutes?

Cheers

Dan

spilegi, 2021-02-16:
@njfaria 
Actually I realized this software does exactly what I want, I just wasn't familiar with mqqt.  It appears to only run on arm, not intel.   Is it something I'm doing wrong? or is it only compiled for arm?

njfaria, 2021-02-16:
it runs only on arm,
a raspberry pi consumes less power than an intel machine 24/24.

mrgram, 2021-03-09:
Hi Nuno! This is awesome, this solved a lot of inconveniences I had. One of the things I'm using it is in nodered to automate if the pack should be charged or not, and at what percentage, at night (low tariff) based on the weather for the next day (% of sun)
I noticed that after a day or 2, I stop receiving mqtt data in nodered. And just ticking the mqtt box in dalylog solves the problem for a couple of more days. Is anyone having this? issue?
Is it possible to comunicate the real SOC to inverter? Im using the pip5048mg inverter.
thanks!

danielforster, 2021-03-09:
Hi, If it helps, some tips I have found when using this really good software:-

1. It wont run on a Pi 0, they seem to be too slow and dont give all the readings to MQTT all the time.
2. On a Pi 4, it works perfectly, like it sings along.
3. DalyLog seems to not like the MQTT link going down, if you are using WiFi on the Pi's, make sure its really strong. On the Pi 4's, I use the LAN wired connection, which is much better. Ditto if the MQTT server stops for some reason, DalyLog seems not want to reconnect, but if closed, then reopened, it runs again perfectly.

Dan

ROQM76, 2021-04-24:
How do i monitor multiple daly bms's? Do i need multiple pi4's?

danielforster, 2021-04-25:
I use one pi 4 per BMS, 4 battery backs so 4 pi 4's. Get one pi working perfectly then duplicate the sdcard for the other pi's...

njfaria, 2021-04-26:
uh, thats quite an investment. you may try to make separate folders. duplicate the folder.
change the prefix in the config separator. it will send to MQTT with a different name.
about the reconnections to MQTT, i am solving that problem if the broker is not available for some reason, to make a reconection.

danielforster, 2021-04-27:
Nuno,

I have 4 seperate batteries and like to keep them all individual. The RPi's cost a small amount compared to the 64 LifPo4 cells, and 4 BMS units. They all work really well together too. I do suffer from the odd MQTT disconnect, and like you say, DalyLog does not reconnect, meaning I remote in (VNC) when this happens and close/reopen DalyLog.

Other than that, its very good software, brilliant in fact.

Daniel

mrgram, 2021-04-27:
Great! 
I didn't think in using 2 instances to monitor more than one pack! I assumed that since they are connected in parallel it didn't really matter, but regarding detailed data its a great trick! thanks!
About the MQTT disconnections I still have them also, but its better since I overclocked my raspberry, happening approximately only once a week now. I use twitter node to notify me when SOC data as stopped being received, and then, also with vnc I reconnect dalylog. Its good news Nuno that you are working on it, thanks!

Thanks to this software I've been able, with nodered, to automate my home consumption to 100% green! 

Before, because the SOC declared by the inverter wasn't correct I had it switching to grid mode during peak hours when in reality I still had a good SOC. One great thing also is the ability charge to a specific % depending on the weather(sun) prevision for the next day.

Thanks again

danielforster, 2021-05-06:
Nuno,

When you do the MQTT reconnect issue, is there any chance of a field in DalyLog to pick the time that the MQTT updates are sent? Maybe in minutes (to keep traffic down I would chose something like 10 mins or so).

Cheers

Daniel

njfaria, 2021-05-08:
@danielforster @mrgram i uploaded a new version which should have the issues with MQTT solved.
a new field box allows to change the update rate.
It will give an error if you copy it over the old version and execute it, cause the config file has changed. Just delete your old config file before starting the software, and reconfigure to your values.

danielforster, 2021-05-09:
Nuno,

Superb :-).

Been testing for 12 or so hours on all four of my battery pack/Pi's and all good. The MQTT update time interval addition is perfect.

Many thanks for your hard work. You are a star...

Take care,

Daniel

danielforster, 2021-05-12:
Nuno,

Quick update. DalyLog stays connected to MQTT perfectly now, no dropping connection at all, so a huge thanks :-)

Take care,

Daniel

spilegi, 2021-05-12:
I can confirm the same.  Great update.  It appears to resolve another issue I was seeing where the first 2 or 3 cell's would sometimes display zero values.

Eumobong, 2021-05-17:
Which communication interface if the Daly bms are you guys using? Uart or CAN?

danielforster, 2021-05-17:
Hi,

UART is what I am using, in fact, I have the BT module plugged into the CAN, the LightBoard plugged in, the the Pi on the UART. All outputs work together fine, while outputting via DalyLog I can still make adjustments via the iPhone app (to the BMS settings) and see the battery level on the light board.

As a test for the updated v1 of DalyLog, my internet went down and came back 6 hours later, DalyLog reconnected to my MQTT server (some 4000 miles distant - via draytek/draytek router tunnel) without delay. 

Daniel

danielforster, 2021-08-31:
Nuno,

A huge thanks for adding he battery status.

Take care.

Daniel

antoncatalin, 2022-01-26:
Hello,
I have one raspberry pi 4 with ICMPI instaled OS from https://iccsoftware.co.za/. I tried to connect to a BMS daly with dalylog but I don't know what the communication settings are. Can you help me?

mrgram, 2022-03-28:
Ola @njfaria, I come back with the issue that some had and may still have, I tried to make it start on boot several ways but I cant   figure it out. Is anyone having this issue? I tried to us alternative methods like crontab, rc.local file, etc... but nothing.
Any help would be much appreciated.

kxn25, 2023-04-04:
Sorry but in the ZIP file is no setup.sh!
Where do i get the files?