njfaria/dalylog #3: Error between main display, and data send to MQTT

firemen027, 2021-03-09:
Hi, 

I recently install dalylog on ma RPI4 with uart to usb cable

The problem is reading from battery 01 TO 06 , they display strange number
Many time battery 1 display 0.0 or 13.xx volts. sometime the correct is value
Other battery 02 to 06 display 0.0 volts or real value
Main display upper part never display this  strange value in the min, max cell
[image]

The output to mqtt also show the wrong value.
[image]

Here example the main page showing good result , but the cell voltage are wrong in the lower part of the display. The value display in the lower part are the one send to mqtt.

I hope I'm clear in my explain, english is not my primary language.

Thank you NjFaria for any help

firemen027, 2021-03-09:
Hi again, for unknow reason  my last picture simply don't show, so here we are
[image]

spilegi, 2021-03-30:
@firemen027

What version of Raspberry pi are you using?  I had the same problem when using a Raspberry Pi 3.  I do not see the issue when using a Raspberry Pi 0 wifi, or a Raspberry pi 4.    I still use the Pi 3, I just coded around it to drop any readings over 5 volts.  It will display the correct voltage some of the times.

firemen027, 2021-03-30:
Hi Spilegi,

I use Raspberry Pi4 8gb. I also have my mppsolar link by usb port on the same PI. Presently I send all data to mqtt, then drop all false data below 2 volts and over 4 volts with nodered, then write data to influx DB. I Also  search a way to receive data to mqtt when Daly is balancing some cell.

I first want to read Daly with some bluetooth link and send it to mqtt, but don't found anything.
I have access to the daly only at the cabin, so everything is a long process.

njfaria, 2021-04-26:
> Hi again, for unknow reason my last picture simply don't show, so here we are
> [image]

trying to simulate that. i dont have a daly bms, trying that on some users who are using it.

ans-gw, 2021-05-28:
Hi njfaria,
compliments for you good job,first....
..and today I notice in last version 1.0 program dont send anymore  CELL TEMPERATURE by Mqtt.
Can see Cell Temperature in Dalylog program...but not in Mqtt.
The past version 0.9 was showing Cell Temp.

Dont know if you take out Cell Temo,or you forgot it...

Thank you

njfaria, 2021-08-25:
@spilegi @ans-gw @firemen027 v 1.0.1 out. bugs fixed

closed 2021-08-25 by njfaria