njfaria/dalylog #5: Source code?

spacie2136, 2021-04-10:
Can we maybe get the source code?

amotl, 2021-04-19:
Dear Nuno,

together with @danb35, @jblance and @firemen027, we are currently having a nice conversation over at https://github.com/daq-tools/kotori/issues/50, https://github.com/jblance/mpp-solar/issues/75, https://github.com/jblance/mpp-solar/commit/128338eda8c7d1ac7ba4f3dafd62e38027109d0e and https://github.com/jblance/mpp-solar/issues/83 in order to support the community with fine software on the topic of energy monitoring with MQTT, InfluxDB, Grafana and Kotori.

If that resonates with you, you might enjoy joining us in order to collaboratively work together on bringing data from the Daly BMS into this lovely ecosystem.

With kind regards,
Andreas.

spacie2136, 2021-04-19:
Actually i'm looking into to doing something similar. But for a wider community i think.

I want to create a simple tool running on raspberry or ESP8266 (or ESP32) transmitting data from BMS to MQTT. Later also changing configurations via MQTT.

I already have my Victron solar inverter sending data to my MQTT server! And also saving data to InfluxDB and displaying via Grafna (Screenshot attached.)

MQTT is easy to handle and can be integrated into Homeassistant for example.
 
[image]

amotl, 2021-04-19:
Hi @spacie2136,

thank you for your response. This is also a nice Grafana dashboard you are using for that purpose. @danb35 showed his at https://github.com/daq-tools/kotori/issues/50#issuecomment-821295561. Do you mind sharing the JSON source of yours?

> changing configurations via MQTT

For that purpose, you might be interested in [1,2].

With kind regards,
Andreas.

[1] https://community.hiveeyes.org/t/using-grafana-for-submitting-downlink-commands/2959
[2] https://github.com/geeks-r-us/mqtt-panel

liebefuroscar, 2021-05-03:
That sound great! I´m looking for buying a DALY  BMS for my new project with LiFePo4 battery and Daly BMS has the Rs485 and Canbus connection :D.
Y need MQTT fot integrate those datas on my Home Assistant and monitorize that.
I have a lot of ESP32´s in my house, but if need to buy an Rpi Zero, i´ll buy it. No problem.
With kind regards,
Oscar.