njfaria/dalylog #6: No data displayed but appears connected.

albuchan, 2021-06-02:
Hi I have a Daly BMS 12v 4S 100amp device. And I have a daly usb-mon (meaning it has the 3 pin monitor connector rather than the 5 pin Uart)  serial converter. It works fine with the  win10 Daly Sinowealth tool v 0.2 using "SH39F003" IC as connection type. However using the same converter on a USB on the Pi. The serial is found (see image) and the green light below start flashes.
but no data received - the databoxes on tab "main" remain empty except SOC 100.  Is there something I have missed doing? Can I check if the right serial is found? I do have a wireless kb/mouse and another USB device for connection to NMEA2000 bus (this is on a boat).

Thanks
[image]

njfaria, 2021-06-02:
Hi,
Do you have the possibility to take a picture of the converter ? The users
i know where using the dalybms software. Thats the protocol i rely on. Why
are they using different software for each bms ?
I found some info on
https://energytalk.co.za/t/daly-smart-bms-connections-llt-bms-communication-with-venus/322
I am reading threw it. Since that is on a boat i presume there is no
internet on the raspberry to that i could connect to threw vnc.

albuchan, 2021-06-02:
Hi Nuno
Thanks for quick reply. There DOES seem to be 2 SW in use by Daly and not much written on Sinowealth for the exchange protocol. Personally I think the uart rx/tx 9800 is the same, but not clear what handshake sequences needs to get the box to reply. Here is the converter I have 
[image]
 ON the DIYSolar forum someone uploaded a english translation of the "manual"... doesn´t talk about protocol more what data can be found.  (here https://diysolarforum.com/threads/daly-smart-bms.9070/page-4) I have downloaded it and attaching it here..
[Sinowealth BMS Tool Introduction V0.1 English.pdf](https://github.com/njfaria/dalylog/files/6585113/Sinowealth.BMS.Tool.Introduction.V0.1.English.pdf).
Access is available to the pi.. but via anydesk. I have a fair amount of other sw running that I wouldn´t want to disrupt. I would need to be on the boat to let you in.. VNC is also on the pi... but since I don´t have static IP i use remote.it as a wrapper. Suggest we use email to exchange any credentials or agree when you could access remote.

Alastair

solarsnoop, 2021-06-02:
[image]
U need this converter for daly solpiplog. The monitor usb adapter is not rs485 protocol

solarsnoop, 2021-06-02:
[image]
And you need this output port on your bms

solarsnoop, 2021-06-02:
I use dalylog with 3 different bms all connected with rs485 it will work fine. Maybe uart is working too but this must comment nuno

albuchan, 2021-06-02:
Hi.  I do have another converter for 485( not Daly one and so connection is tricky since I don’t have a 5 pin connector that fits . But since none of the other ports are marked I can’t be sure that it does or doesn’t have a 485. My dealer told me that this model does not have 485. What I will try to do is connect on win10 with sinowealth and see if I can snoop the rxtx exchanges. Sorry for bad photos but installation doesn’t let me take photos easily

Temp sensor and Bluetooth

Uart serial + ????

Alastair

njfaria, 2021-06-02:
feel free to send me a mail to solpiplog at gmail.com regarding access to
your pi.

albuchan, 2021-06-04:
Nuno, I have been looking into this, there are quite a few posts confirming the info you shared in the energytalk forum that there are 2 daly BMS tools -  - In particular tanoshimini describes in detail the Daly BMS PC software using 486 protocol used  https://diysolarforum.com/threads/decoding-the-daly-smartbms-protocol.21898/ - I am guessing this is the same protocol you are using in your pi app. However, in https://diysolarforum.com/threads/daly-bms-rs232-communication-protocol-crc-calculation.21853/ mbeq99 is focusing on the Sinowealth BMS tool which is 232 based and seems to be what works on my BMS. I also "sniffed" the com port and saw the 3 byte sentences. It also appears that Daly are not being very forthcoming with documentation about this protocol as they have been with SmartBMS tool. I think that it would need a complete new comms module in your code to work, and I am not sure there is enough info to do this (I am not a programmer even though I have a good background in electronics and IT). So I think best I can do is thank you for replying to me and for the work you have done.