njfaria/dalylog #7: MQTT solpiplog/daly/temp1 not send?

SunnyPerth, 2021-06-10:
Hi Team, 

running the latest version (1.0.0) of this fine piece of software, have been running it for the past few weeks and it works like a dream. Love to get the real data from the Daly into my various dashboards. 

However it seems that the tool doesn't send the /daly/temp1 mqtt topic? Is that correct? 

Cheers,
Yuri

danielforster, 2021-06-16:
Nuno,

I just noticed that the battery temp sensor is not in the MQTT package sent too. Can you see if you can sort this?

Many thanks as always,

Daniel

njfaria, 2021-06-16:
Yes some user mentioned that. A bug that i implemented. Do i have access to
the raspberry ? I dont have a daly...

danielforster, 2021-06-16:
Nuno,

Yes, I have 4x separate 5.5kw battery packs with a pi4/BMS in each pack. They all report to MQTT as pack1/ pack2/ pack3/ and pack4/.

I can SSH into each pi or use VNC to see each pi's desktop.

Take care,

Daniel

SunnyPerth, 2021-06-20:
Hi Nuno, 

Any luck with this? If you don't have access to a Pi and Daly, I do (and so does Daniel I think).  
As the application on the pi does show the temperature, I know that your app does have the data. 
So, I think it's as simple that your MQTT routine doesn't send the data, it probably just misses the line? 

As you do not share the source code, I can't check / help. But I'm happy top help troubleshoot and test. 
Let me know if you can share the source code, or a version that you have fixed and I can test before we publish it? 
Happy for you to send me a pm to solve. 

cheers,
Yuri

SunnyPerth, 2021-08-31:
Super Thanks Team!

SunnyPerth, 2021-09-03:
Hey Team, I've installed the new version, and now get the temp - great. 
However I'm now seeing an odd behaviour that I can't explain in home assistant, see also image attached. Just wondering if others see this or have seen this, and since the only thing I've changed is the Daly software if this is related due to the upgrade. 
So what I see is that opening sensor data detail graph that I get via the Daly BMS, the line chart is now coloured in above/under a (to me) value. 
See the SOC and remaining ah examples. [image][image]

I've not (yet) reverted to the old version to see if it's due to the new version - will do in next days, but wondering if anyone else has seen this ? 

Cheers

njfaria, 2021-09-03:
How did it looked before ?
I dont have a daly, therefore i am relying on your testing.

SunnyPerth, 2021-09-03:
Thx mate, appreciate you looking into this. This is how a graph should look like. [image]

It's almost like the app is sending a constant value and then the real value - I'll have a quick look this morning to see if that's the case & if this behaviour stops when switch back to previous version.

SunnyPerth, 2021-09-04:
Hi mate - just checked. The latest version of the app 1.0.1 does send mqtt values alternating, the (wrong) initial value and the real value, followed by the wrong and then real value.. this results in the graph looking likes it's filled in, but if you would zoom in you see both values. 
It also does send the data that every second ( it every minute ). 
I've reverted to previous version which solved the problem, but didn't/doesn't show the temp ;)

njfaria, 2021-09-04:
For some reason do you got two versions of dalylog logging ?
I am trying to get some feedback from other users.

à(s) 01:28: