njfaria/dalylog #10: Where is the executeable file?

derwojna, 2021-09-20:
Hello,
I am new to Github, so forgive me this stupid question: Where do I find an executeable file to start "Dalylog"?
Since I cloned your url to my Raspi, all the files appear, but no executable file nor a setup.sh or similar. What do I do wrong?

njfaria, 2021-09-21:
sorry for delayed answer
the executable is under releases
[](https://github.com/njfaria/dalylog/releases/tag/1.0.1)

derwojna, 2021-09-24:
Thanks for that. I started the application but no device is found although my DalyBMS is attached to the Pi via USB-Cable. Are there other instruction to follow to make the Pi "see" the BMS? "lsusb"-command showed nothing. I found no new device in /dev. Perhaps you know what I am doing wrong?

kxn25, 2023-05-02:
How can i auto start dalylog on my RPI?