njfaria/dalylog #18: Dual dalylog and BMS

Pinky-bit, 2023-11-22:
Hi all, is it possible to use two Daly Smart BMS at the same time and send data to emoncms with different prefixes? Thank you

solarsnoop, 2023-11-22:
Just change the prefix to example solpiplog2 that's all. Than u find the data's in your mqtt broker under solpiplog2/daly

Pinky-bit, 2024-01-07:
> Just change the prefix to example solpiplog2 that's all. Than u find the data's in your mqtt broker under solpiplog2/daly

Ok, the data is seen by the mqtt broker, but I can't import this data into emoncms. I also tried different input methods in emoncms. It only sees the first bms.

kxn25, 2024-01-23:
Have you made a copy of the Folder to run 2 instaceses?

Pinky-bit, 2024-01-23:
> Have you made a copy of the Folder to run 2 instaceses?

Yes!

Pinky-bit, 2024-01-24:
> Thank you.

Were you able to publish data from both bms on emoncms?

kxn25, 2024-01-24:
iam not using emoncms.
And i have not orderd the second battery pack, just checking if posible.

kxn25, 2024-02-02:
How can i setup daly log in autostart? sysctrl services?

Pinky-bit, 2024-02-05:
> How can i setup daly log in autostart? sysctrl services?

In my Raspy I copy it to /home/pi/.config/autostart

kxn25, 2024-02-06:
Which of these files must i copy in autostart?
[image]

Pinky-bit, 2024-02-06:
In autostart I copied the dalylog.desktop, file that I don't see in your directory.

kxn25, 2024-02-06:
And how do i get this file?

Pinky-bit, 2024-02-06:
> And how do i get this file?
> can you send me yours? 

You can find it inside the dalylog_0.9.1.zip file

kxn25, 2024-02-06:
Thank you i will try that.

kxn25, 2024-02-20:
is it possible to get errors from the BMS?

kxn25, 2024-04-10:
> Just change the prefix to example solpiplog2 that's all. Than u find the data's in your mqtt broker under solpiplog2/daly

The SOC is always initalise between the two. I have disabled the second SOC switching.
Is that normal? and will it switch the one i have enebaled?