njfaria/dalylog #19: Switching Point SOC?

kxn25, 2024-03-14:
Hello,
is it possible to make the Switchpoint according to the Cell Voltage and not the SOC?
Since the SOC is from time to time drifting apart and not accurate any more!

Does some one have a better solution to make Solpiplog to Switch from Grid to Battery and back?

kleinegitty, 2024-12-28:
> Hello, is it possible to make the Switchpoint according to the Cell Voltage and not the SOC? Since the SOC is from time to time drifting apart and not accurate any more!
> 
> Does some one have a better solution to make Solpiplog to Switch from Grid to Battery and back?

Hi kxn25, you helped me last summer with the AS608 tasmota fingerscanner, but can't see your posts anymore....
Where are they? Thank you!

kxn25, 2024-12-29:
Hello @kleinegitty here is the link

`https://github.com/arendst/Tasmota/pull/9568`